ArduPilot/UAVLogViewer #171: port to upstream mavlink js

Williangalvani, 2020-03-29:
upstream recently had mavlink2 support incorporated, we should try to use that to take advantage of the new messages

davidbuzz, 2020-03-29:
Re the updated generated .js mavlink parser ...although it now can parse mavlink2, I found it lost the ability to parse mavlink1 in the process,  so I embedded two variants, an older as v1, and a newer as v2 when integrating it into mavcontrol web GCS.

Williangalvani, 2020-03-30:
Hey, thank you for the info! I'll take a good look at the state of things when I do it.

I ended changing a lot of the original code I used(mostly for eslint to accept it), and I think it supports both v1 and v2 right now. I didn't try to merge things upstream because I wasn't confident of what I was doing, haha.

Williangalvani, 2021-08-30:
fixed by #298 I did keep my ould parser though, as that should be fine with mavlink 1 and 2 (but no signing).

closed 2021-08-30 by Williangalvani