ArduPilot/UAVLogViewer #175: support streaming data

Williangalvani, 2020-03-30:
We can update the plots like [this](https://plotly.com/javascript/streaming/).

The data will likely come from patrick's [mavlink2rest](https://github.com/patrickelectric/mavlink2rest).

davidbuzz, 2020-03-31:
MAVControl ( https://github.com/jabelone/MAVControl ) has a number of backends python, fat-node.js, and a 'thin one' (also node.js ) , which is really just a mavlink-over-UDP to mavlink-over-websocket stream using SocketIO websocket, with the mavlink parsed 100% in-browser. I was thinking of implementing something that would allow TLogViewer to render 'live ' data, and that might be it.

Williangalvani, 2020-04-01:
That sounds interesting too! I'll take a look!
I haven't thought about live data at all while building the viewer, so I'm afraid we'll have to change a lot of things along the way.

davidbuzz, 2020-05-14:
http://smoothiecharts.org/builder/ is also cool as a live-streaming graph.