ArduPilot/UAVLogViewer #184: Add button to align zero on all plots and autoscale to largest one

Hwurzburg, 2020-06-11:
this wont help much in mixed plots, but I have been running a lot of rangefinder tests which compare three alts and this would have saved me soooo much time aligning and scaling everything so that they are the same...even then the scale granularity would not allow me to perfectly align the three altitude plots...baro, rangefinder, rangefinder target....this would be great for checking attitudes , pids, etc. also...and if you had to have mixed graphs, it would be nice if the scales/zeros of the previously aligned graphs do not get adjusted when you add new non-homogenous plots...and it would screw up plots that dont have zero, but the user would just have to be cautious in pushing the button

Williangalvani, 2020-06-15:
So you need the same zeros but different scales? One alternative (not great) would be to just multiply some of them and use the same axis

Hwurzburg, 2020-06-15:
no same zeros....same scales...
[image]

Williangalvani, 2020-06-15:
the dropdown menu with a number selects with "axis" the plot is at. In this last example you could put all plots at axis 0:
[image]

Hwurzburg, 2020-06-15:
brilliant!...this really needs a wiki!

closed 2020-06-15 by Hwurzburg