ArduPilot/UAVLogViewer #185: Make messages visible at least up to cursor..

Hwurzburg, 2020-06-11:
preferably always show all messages in the list...syncing to the timeline cursor is okay, but unfortunately, the messages stop in time well before the cursor...should at least be a few in the scroll list beyond the cursor time stamp...

Williangalvani, 2020-06-15:
what about some visual feedback? make the ones up to the cursor bold, maybe?

Hwurzburg, 2020-06-15:
that sounds fine, but the big issue, is that you cant see messages all the way up to the cursor point...they stop before that

Williangalvani, 2020-06-15:
Right! I missed your point. Can you share a log where that happens?

Hwurzburg, 2020-06-15:
I will try  to find the one I was having problems with....

Hwurzburg, 2020-06-18:
cant find the logs this was doing it...so will close until I see it again...

closed 2020-06-18 by Hwurzburg