ArduPilot/UAVLogViewer #187: Wrong time shown in Cesium player

Hwurzburg, 2020-06-15:
https://www.dropbox.com/s/p4ut300nu3mu2ok/2020-06-14%2015-55-06.bin?dl=0

log was taken on June14....player shows Jul 6 2020

Williangalvani, 2020-06-15:
any chance it was actually June 15?

Hwurzburg, 2020-06-15:
[image]

Williangalvani, 2020-06-15:
I mean the date you flew it. do you happen to remember the aproximate time as well?

Williangalvani, 2020-06-15:
does this look right?

Hwurzburg, 2020-06-15:
Its in the log...yesterday .....6/14/2020 3:53PM local time

Williangalvani, 2020-06-15:
Right, my bad! I need to check how cesium time works

Williangalvani, 2020-06-16:
Best I can do for now!
[image]

Unfortunately adding localization to Cesium is going to take some time... I'll have to:
 - guess timezone from gps (found a lib for that)
 - override some internals of cesium to render the localizated string

Hwurzburg, 2020-06-16:
UTC is fine...just having stuff in the future was a bit odd before!

PR #188 (2020-06-17): DataflashParser: Fix gps time calculation
Fix #187
closed 2020-06-16 by Hwurzburg