ArduPilot/UAVLogViewer #189: Hang on Plane PID tuning preset when its a quadplane with fixed wing and vtol modes

Hwurzburg, 2020-06-16:
not sure what this preset tries to the look at (will try to find a plain plane log tomorrow), but running it on a quadplane log , it hangs indefinitely

Williangalvani, 2020-06-17:
I just tried the log from autotests and it seemed ok:
http://autotest.ardupilot.org/ArduPlane-log.bin

can you share the problematic one?

Hwurzburg, 2020-06-17:
https://www.dropbox.com/s/qmxrrvns3l2zffw/2020-06-13%2009-59-20.bin?dl=0

btw, it wont load properly on plot.ardupilot.org....it runs on tlog.ibexcps.com....it hangs on plan pid tuning presets...

Williangalvani, 2020-06-17:
what about plotbeta.ardupilot.org?

Williangalvani, 2020-06-17:
it might have been fixed by https://github.com/ArduPilot/UAVLogViewer/commit/7e8278e73f2fe22e7502ce155d027540c2c7dada

Hwurzburg, 2020-06-17:
loads on plot now, but still has infinite hang when selecting preset plane->pidtuning->rol or pitch controller

Williangalvani, 2020-06-17:
both presets are fine here...
tested in chrome 80.0.3987.122 and firefox 74.0
[image]

Make sure you refresh the browser so it loads commit ad5988

Hwurzburg, 2020-06-17:
in Linux on Firefox and Chromium, it does not hang....on Win 10 Chrome Version 83.0.4103.106 (Official Build) (64-bit)  it does

Williangalvani, 2020-06-17:
damn windows. Mine has been misbehaving a lot. I'll try to test it later

Williangalvani, 2020-06-17:
[image]

83.0.4103.97 here...