ArduPilot/UAVLogViewer #218: GUID.vX unit is (cm/s) in Ardupilot log rather than (m/s)

cnpcshangbo, 2020-08-25:
https://github.com/ArduPilot/UAVLogViewer/blob/82cdbd4660347d9f55e9602750530c32ab5349d4/src/assets/logmetadata/copter.xml#L1951

[image]

Refer to Ardupilot code:
https://github.com/ArduPilot/ardupilot/blob/5fad6ddddbdffa2f3135ba1b8960614f4d87d374/ArduCopter/Log.cpp#L434

Williangalvani, 2020-08-25:
Hi. can you share this log?
The values here should actually be transformed to meters (the parser reads the multiplier field and applies it).

cnpcshangbo, 2020-08-25:
@Williangalvani Sure. You can assess the flight log [here](https://drive.google.com/file/d/1smGmbk2T0EFLbz4937fqVqIHgbxrgY5X/view?usp=sharing). I created it with SITL and sent the command with a DroneKit script.

Williangalvani, 2020-09-06:
Hi, I managed to replicate this. It was a bug introduced a while ago when I re-structured the messages in memory to save ram. Thank you for reporting it!