ArduPilot/UAVLogViewer #222: Plotting discontinuous dataflash records and GPS time

kd0aij, 2020-09-06:
Some log records aren't recorded continuously; there may be intervals of time when no samples are present.
mavproxy.py shows this by plotting a symbol at each time for which a sample exists, and connecting the symbols with a solid line.
[image]

but UAV logviewer just draws the line, and obscures the fact that some samples are missing:
[image]

It would also be nice to have the option to display the GPS time instead of time since boot.

Williangalvani, 2020-09-06:
Hi, can you share this log?

your time axis seems weird, what commit hash and build date do you see under the dropzone?

kd0aij, 2020-09-06:
I think I generated those plots using plot2.ardupilot.org:
Commit a2483f
Built Sat Jun 13 2020 01:35:54 GMT+0000 (Coordinated Universal Time)
Link to the log: https://drive.google.com/file/d/1EDP0BTSfkGiHtmKRHWNSj604GnuNI068/view?usp=sharing

Williangalvani, 2020-09-06:
> plot2.ardupilot.org

Please use plot.ardupilot.org or plotbeta.ardupilot.org. I'll try to make plot2 redirect there.

I am confused here... Should there not be a new point if the value hasn't changed? We still received a value even if it hasn't changed...

kd0aij, 2020-09-06:
Thanks, didn't realize my link to stable was stale.
This is interesting; looking at the end of the RLX record in that log with stable. It shows the timestamp for the last record, and no line is plotted beyond that:
[image]

But there are also gaps in the sequence of RLX records in the middle of the log, even though it doesn't appear that way on the plot.

Williangalvani, 2020-09-06:
I will leave this here for future references:
[image]

> This is interesting; looking at the end of the RLX record in that log with stable. It shows the timestamp for the last record, and no line is plotted beyond that:

My guess is that that was the last message received. it seems to match what I get in mavexplorer. I don't understand the reason for the discontinuities, or, even worse, why our RLX.tz changes before mavexplorer's.

Well, let's see what I can find...

kd0aij, 2020-09-07:
Yes, the plotter should definitely not extrapolate values either forward or backward in time.
And assuming there's no bug in mavexplorer (and its plots match my expectations for the RLX record),
it will be interesting to see how values are getting extrapolated into the past.

The RLX records are written only during the interval from entry into qloiter mode to raising the throttle here, and it would be confusing to see the values change outside of that interval (I am using the RLX record to test a code change):
[image]

Williangalvani, 2020-09-07:
Hah. it turned out to be typo. I blame the english language!

Williangalvani, 2020-09-07:
@kd0aij do you mind checking how it looks now in http://plotbeta.ardupilot.org?

kd0aij, 2020-09-07:
Looks correct to me now; I can tell that RLX stops when it should by hovering with the mouse, but I'd prefer that the line was also suppressed, since the continuous ramp from one value to the next gives a false impression without sample markers.

[image]

Williangalvani, 2020-09-07:
> which is clipping the text at top and bottom of the plot window, and drawing text on top of Y-axis lines at the left.

This is due to plotly's implementation. It doesn't use regular css, instead everything is placed in a [-1,1] interval. That makes it very hard to match different window sizes. For now it works well on 1080p here, and is show be at least usable on lower ones.

Unfortunately, as you see in #223, enabling markers is VERY expensive without scattergl, and for now scattergl is not compatible with the rangeslider, so that will have to wait. Meanwhile I merged #228 and it should be live in plotbeta.ardupilot.org when CI is done.

kd0aij, 2020-09-07:
I see that the text placement is OK if I expand to full screen, so that's OK.
And given that performance is poor with markers, the hover text is sufficient to resolve any ambiguities for the present.

kd0aij, 2020-09-18:
samples are now connected by a straight line in beta, so gaps appear as long, straight lines and hovering with the mouse can be used to identify gaps in the data

closed 2020-09-18 by kd0aij

Williangalvani, 2020-09-18:
Hi @kd0aij ,

It should actually look like this, now (in http://plotbeta.ardupilot.org):
[image]

kd0aij, 2020-09-19:
that's great! Didn't know you were still working on it. Now I'll have to see if I can do the same for mavexplorer.