ArduPilot/UAVLogViewer #229: Record animation to a video file

kd0aij, 2020-09-15:
On Windows 10, I use the "Xbox game bar" to record segments of the animated vehicle display.
It would be nice if the viewer could export directly to a file instead.

chennighan, 2020-11-13:
@kd0aij I recently started working on this. It's coming soon, just polishing it up, but this will enable you to record the screen and download the file once finished.

kd0aij, 2020-11-13:
@chennighan many thanks, that will be quite useful for communicating the results of log analysis

magicrub, 2020-11-13:
Would be cool if there was a "publish on YouTube" button to save the user a few steps. Would require them logging into Google and such.

chennighan, 2020-11-13:
@magicrub I propose we add that enhancement as a new issue with the addition of implementing the Google Sign-In and Youtube APIs.