ArduPilot/UAVLogViewer #231: Mavlink version

auturgy, 2020-09-29:
Generated mavlink.js is out of date and does not appear to be built from the ArduPilot mavlink repo.

Williangalvani, 2020-09-29:
This is a duplicate of #171

closed 2020-09-29 by Williangalvani

auturgy, 2020-10-13:
Not actually a duplicate: 171 relates to the parser, I'm referring to the xml source and version.
Due to differences between mavlink/mavlink and ardupilot/mavlink, this needs to be built from the xml in ardupilot/mavlink
@amilcarlucas is doing great work trying to close the gap, but we're not there yet.

Williangalvani, 2020-10-14:
Right! That was actually bad wording from my side. That is what I meant. I renamed that issue. It is going to take a while to get it done as I have done quite a few changes in there, including a pre-parser so we only parse the required messages when they are required.

I'll try to give this another go this weekend.