ArduPilot/UAVLogViewer #232: can you make a artificail horizon with speed and altitude like on mission planner?

Xarin94, 2020-10-03:


AndKe, 2020-10-28:
yes please :+1:

Williangalvani, 2020-11-03:
we can probably use this. https://github.com/uCibar/vue-flight-indicators

AndKe, 2020-11-03:
yes !  that would be perfectly readable.
The altimeter may need to be scaled to the levels we often fly, but the altitude is not necessarily the most important to visualize.

Williangalvani, 2020-11-10:
@AndKe @Xarin94 the attitude indicator widget should be available in http://plotbeta.ardupilot.org in a few minutes.

Are the others needed at all?
Heading
Vertical speed
Air speed
Altimeter

Xarin94, 2020-12-02:
perfect thanks

AndKe, 2020-12-02:
@Williangalvani 
When analyzing logs, I often watch closely IAS/altitude and maybe other instruments in the critical moments.
One can always plot lots of data just to see them, but a plot is not as intuitive to read as standard instruments. (scale can vary, there can be lots of other data plotted, to reveal the cause .. cluttering the display.)
Penultimately, a typical PFD would be best. - but that is a bigger request :smile: