ArduPilot/UAVLogViewer #245: Altitude Shown

liquidman864, 2020-11-12:
Can you please confirm the source of the altitude in the log viewer? Is it from GPS? or Baro? Or the reported aircraft altitude? Just trying to confirm which altitude we are actually seeing on the 4d plot. Thank you!

Williangalvani, 2020-11-12:
it depends on which position source you have selected in the 3d tab, and which kind of log you are analyzing.
For dataflash logs, the position can be taken from either POS, AHR2, or GPS messages.
For Mavlink logs, the position is taken from GLOBAL_POSITION_INT, GPS_RAW_INT, AHRS2, or AHRS3.

The position of the first valid point is then tied to ground at that position (which depends on the cesiumjs terrain data).

What behavior are you experiencing, what were you expecting instead?

liquidman864, 2020-11-12:
Thanks for the quick response. We are looking at a Dataflash log (Ardupilot .BIN file). Trying to make sense of some disrepencies between our GPS, Baro, and our actual altitude. Could you possibly point us to some more info on how the POS and AHR2 values are generated?

Williangalvani, 2020-11-12:
we dont change them much: https://github.com/ArduPilot/UAVLogViewer/blob/master/src/tools/dataflashDataExtractor.js#L248

you could look in ardupilot source where they are sent. for example AHRS is sent [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/LogFile.cpp#L453)

There's also some documentation [in the docs](https://ardupilot.org/copter/docs/logmessages.html)

kd0aij, 2021-12-23:
Is there a way to input an altitude offset to make the trajectory at takeoff/landing match the runway elevation?