ArduPilot/UAVLogViewer #257: Display a "ribbon" for flight path (enhancement request)

kd0aij, 2021-02-22:
A ribbon also conveys the roll angle, making the flight path display more informative. 
Example from flightcoach.org which also uses color to indicate top/bottom and port/starboard:
[image]

Williangalvani, 2021-02-22:
oh, I like this one! implementing it may be annoying, though..
should we allow switching between this color coding and the flight modes?

kd0aij, 2021-02-22:
Perhaps we could always use one color for the bottom, and mode-specific colors for the top, or mode-specific color for top and its complement for the bottom. Of course the RC pattern guys will always be in Manual mode anyway...