ArduPilot/UAVLogViewer #262: Subtraction doesn't work

durka, 2021-03-12:
This error message is a little gaslight-y: 
[image]

Workaround is to use addition instead of subtraction:
[image]

durka, 2021-03-12:
Sometimes you don't get an error message, but the expression involving subtraction just doesn't show up on the plot.

durka, 2022-03-30:
Would you mind reopening this, or should I open a new issue? https://github.com/ArduPilot/UAVLogViewer/pull/306#issuecomment-1082519873

Williangalvani, 2022-03-30:
let's reopen it

Williangalvani, 2022-03-30:
it seems to work if you try plotting `ATT.DesPitch - ATT.Pitch` (with spaces). The regex probably needs some tweaks

Williangalvani, 2022-03-30:
Can you verify that this new fix works? It should be live in http://plotbeta.ardupilot.org after this [action](https://github.com/ArduPilot/UAVLogViewer/runs/5748500629?check_suite_focus=true) finishes. You should probably do ctrl+f5 to ignore the browser cache

PR #306 (2021-10-04): Plotly: use replaceAll when plotting expressions
fix #262
PR #330 (2022-03-30): Regex fix
the "field' group for regex was matching '-',  which breaks subtractions.

hopefully fix #262 this time.
closed 2022-03-30 by Williangalvani