ArduPilot/UAVLogViewer #265: YAW Orientation display

NuvemUAV, 2021-03-26:
Hi guys,

We use our FC mounted -90 degrees on Z axis.

It seems the LOG viwer doesn't check this parameter to plot the drone on the map:

[image]

Williangalvani, 2021-04-05:
Hi, that is because there's only one 3d model for quads and the FRAME_TYPE param is not bein taken into account. I'll try to get to this soon.

closed 2021-04-06 by Williangalvani

Williangalvani, 2021-04-06:
@NuvemUAV would you mind testing https://plotbeta.ardupilot.org/ ?

NuvemUAV, 2021-04-09:
Works perfectly @Williangalvani 

Thanks!

[image]

PR #276 (2021-04-06): Add 3d models for Quad+ and QuadX
Fix #265 

X:
[image]

Plus:
[image]