ArduPilot/UAVLogViewer #266: tkoff / landing altitude not tied to ground

teodoryantcheff, 2021-03-29:
[image]

Trajectory from an arduplane .bin dataflash log. There's a constant offset for the entire flight.
[Here's a link](https://drive.google.com/drive/folders/1FLJNY_BEkBr61HaWhL3NJ2kpd6Dfq9cj?usp=sharing) to the .bin file.

Is there a way to "clamp"  to gound or offset altitude reading?

teodoryantcheff, 2021-03-29:
Changing the trajectory source does not help it one bit.

teodoryantcheff, 2021-03-29:
Just to illustrate the point more clearly:

[image]

Williangalvani, 2021-03-31:
@teodoryantcheff can you try http://plotbeta.ardupilot.org now?

Williangalvani, 2021-03-31:
by "now" I mean after https://github.com/ArduPilot/UAVLogViewer/actions/runs/706607274 finishes

closed 2021-03-31 by Williangalvani

teodoryantcheff, 2021-04-01:
@Williangalvani Perfect! Thanks!

PR #272 (2021-03-31): Cesium: only tie start position to ground height
Previous behavior was checking all points and making sure nothing goes underground.
That is no longer necessary as we can now make the globe transparent.

Fix #266