ArduPilot/UAVLogViewer #289: hangs on ESC fields

mattdholmes00, 2021-07-18:
Viewer is hanging when trying to access any ESC fields.  [Log file example.](https://www.dropbox.com/s/ds5qxecbu73f9te/20210718_00000031.BIN?dl=0)

Williangalvani, 2021-07-23:
Hi, thank you for reporting this!

There are two issues here. the first one is that we actually freeze instead of returning an error. The second one is that we assume the instance numbers start at zero, while in this log they are from 5 to 8 (or something like this)

closed 2021-07-23 by Williangalvani

Williangalvani, 2021-07-23:
The changes are up at http://plotbeta.ardupilot.org
[image]
[image]

mattdholmes00, 2021-08-29:
thanks, this is resolved for me

PR #291 (2021-07-23): Deal with isolated instances, deal with failures loading bad messages.
fix #289