ArduPilot/UAVLogViewer #309: One Click log review from an email or similar

paularthur, 2021-10-17:
Enhancement request :-
Allow emailing of URL that will start UAVLogViewer and load a shared log file in say Google Drive or similar
Benefit - Share missions with less technical folks... " Click this " ..." Press Play..." rather than download this, now drag the log file...

Williangalvani, 2021-11-04:
I have a backend planned. It will be able to host files and open then with a specific "state" of the viewer

paularthur, 2023-07-29:
Hi, what if i make the .bin file publically available (cloud, or whtever) as long as the browser can pull my shared file...

https://plot.ardupilot.org/#/?file=https://drive.google.com/file/d/1ZbKZz87Zw2zBMQV_I5f3HYQthFGOxakv

in this case the file is shared publically and small - but if it was larger, throw up a warning - eg 100MB file ~3 minute download... proceed ?

no need for back end hosting of potentially large .bins

Thanks - this tool is excellent.

Williangalvani, 2023-07-29:
> Hi, what if i make the .bin file publically available (cloud, or whtever) as long as the browser can pull my shared file... [...]

Hi @paularthur ,

that's exactly how it works right now, give it a try =]

except it needs a "direct link", so google drive is unlikely to work, as you need to click the "download" button

paularthur, 2023-08-19:
Thanks - "direct link" sounds perfect - do you have a working example, and what network file share protocols or syntax to make it work (or point me at some doco) thanks again. Im not sure what you mean by "direct link" to an internet shared .bin file.