ArduPilot/UAVLogViewer #319: Graph display never ends

msfujino, 2021-11-22:
I use Mission Planner to download a .bin file from flash ROM and analyze it with UAVLogViewer. 
However, sometimes, the graph display never ends. 
Is there a problem with the .bin file? 
Is there a way to find out?

[image]

Williangalvani, 2021-12-18:
there should be an error in the browser console with more information

msfujino, 2021-12-19:
I have tried "Google Chrome" and "Microsoft Edge" as browsers, but the graph display never ends. 
Attach the bin(ZIP) file. I would be happy if you could confirm it.

[1 1970-01-01 9-00-01.zip](https://github.com/ArduPilot/UAVLogViewer/files/7740137/1.1970-01-01.9-00-01.zip)

Williangalvani, 2022-03-31:
It turned out that your log is missing the MODE message, and the code was breaking while trying to load mode change information. it should work fine now in http://plotbeta.ardupilot.org

closed 2022-03-31 by Williangalvani