ArduPilot/UAVLogViewer #351: Build error

danilopmaciel, 2022-09-14:
WARNING in 4926b2617dd1e77809f6.worker.js contains invalid source map

When execute npm run build

danilopmaciel, 2022-09-14:
when a i run build, i can access from localhost but not access on lan. On lan the page opens, but not open .bin.

cbjornram, 2022-12-06:
Hi danilopmaciel,

I just saw this and I think I had the same issue. When you access it on lan and you try to upload a .bin file, does it get stuck in the loading animation? If so then you have to replace the Cesium Ion token with your own token:

1. Go to https://cesium.com/ion/tokens?page=1, create an account, and create a new token for your UAVLogViewer build
2. Replace the token at https://github.com/ArduPilot/UAVLogViewer/blob/2412d3ab2c9b27e2ef001bcb0b3804d1bd281338/src/components/CesiumViewer.vue#L55-L56 with your own token. Split your token into two lines like in the original, because if one of these lines exceed 120 characters the build will fail. Also make sure to not have any trailing spaces on any of the lines since this will also result in a failure.
3. Build/run UAVLogViewer and try to access again. Now it should work.

Hope it helps!