ArduPilot/UAVLogViewer #365: Does not display the map

markskyyy, 2022-12-12:
Does not display the map
[image]

Cassio-Arys, 2022-12-13:
Same problem here, some logs open but I don't know what changes to these

Davidsastresas, 2022-12-21:
I was just going to open an Issue when I saw this. Any idea where it could come from? I don't mind looking into it, but if any developer familiarized with this codebase has any hint, like it could be this plugin or this library that got out to date, etc etc would be very useful. thanks!

markskyyy, 2022-12-21:
Sample file is working and .tlog files is working. Only .bin not working. Later I will send You a log. Thank You.

markskyyy, 2022-12-21:
I have try chrom and opera

Williangalvani, 2022-12-21:
> Sample file is working and .tlog files is working. Only .bin not working. Later I will send You a log. Thank You.

sounds like we're failing to extract the GPS information, then. I'll wait for the log.
btw, is http://plotbeta.ardupilot.org working? I have a vague memory of working on some of these parts of the code recently.

Davidsastresas, 2022-12-21:
Thanks for the fast answer William.

I just tested here  http://plotbeta.ardupilot.org and it works correctly, map is displayed and vehicle correctly place. However the same log does not reproduce the map or vehicle over it: https://plot.ardupilot.org/

So the same log works on the beta one but not in the non beta one.

Thanks!

markskyyy, 2022-12-21:
Yep. http://plotbeta.ardupilot.org/  is working perfect

https://plot.ardupilot.org/ is not working with the same log file.

markskyyy, 2023-01-09:
http://plotbeta.ardupilot.org/ has many bugs.
1. Plane don`t show roll pitch yaw.
2. after start play log you can`t add more graphs
[image]

markskyyy, 2023-02-19:
Still no any changes?

Williangalvani, 2023-02-20:
hi @markskyyy , I missed your previous message. could you share that log file?

Williangalvani, 2023-02-28:
@markskyyy can you share the log file?

Enggmaug, 2023-03-02:
Hi, 

I have the very same issue. Also works fine with beta, though.

See log file [here (link valid for 1 week)](https://wetransfer.com/downloads/5b86f9f016563170a1e9852ad0518dfb20230302125625/94c818737285e08698ca438de494d06720230302125907/4ff531).

Thanks,
G.