ArduPilot/UAVLogViewer #368: Attitude ( both in widget and in 3D view ), not working in the current beta

Davidsastresas, 2023-01-04:
On the current beta:

https://plotbeta.ardupilot.org/#/

The attitude is no working anymore. If I recall correctly I used this within the last 2 weeks and it worked correctly back then, in case something could have gotten into recently that could affect it.

At some point I would love to get familiarized with this project and contribute myself, but until I am familiarized enough the best I can do is to report the issues. Thank you very much for the time!

Davidsastresas, 2023-01-04:
Sorry, I just realized the default attitude source was set to some XKF source instead of ATTITUDE one. It is working in the 3D representation now.

But the images are not loading for the widget. I see the frame moving around making sense with the attitude, but the image itself is not rendered. I am using google chrome on ubuntu 20.04.

Williangalvani, 2023-02-20:
plotbeta had a lot of dependencies updated, which probably ended up breaking this along with a couple of other things. I'm cheking it now

PR #372 (2023-02-20): use file-loader for svgs
fix #368
closed 2023-02-20 by Williangalvani