ArduPilot/UAVLogViewer #395: Stuck in loading page and unable to view log

teojy98, 2023-06-12:
[image]

Grylls24, 2023-06-12:
I've got the same problem...

FlyingSandal, 2023-06-12:
I have the same problem as well, just spinning but cannot review logs...

gediminasgu, 2023-07-06:
I ran version 1.0 and stuck with the same screen. In the Chrome console, I see errors loading Celsium resources:
[image]

After removing the "waiting" layer, the log seemed loaded. The graph is working, but not a map.

gediminasgu, 2023-07-06:
OK, I was able to fix to by replacing default token with my own here: https://github.com/ArduPilot/UAVLogViewer/blob/master/src/components/CesiumViewer.vue#L56