ArduPilot/UAVLogViewer #402: Error on running npm install

adityaomar3, 2023-09-10:
npm ERR! code 1
npm ERR! path C:\project\UAVLogViewer\node_modules\deasync
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c node ./build.js
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@18.15.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.8.10 found at "C:\Users\Omar\AppData\Local\Programs\Python\Python38\python.exe"
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS msvs_version not set from command line or npm config
npm ERR! gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
npm ERR! gyp ERR! find VS checking VS2022 (17.6.33815.320) found at:
npm ERR! gyp ERR! find VS "C:\Program Files\Microsoft Visual Studio\2022\Community"
npm ERR! gyp ERR! find VS - found "Visual Studio C++ core features"
npm ERR! gyp ERR! find VS - missing any VC++ toolset
npm ERR! gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
npm ERR! gyp ERR! find VS looking for Visual Studio 2015
npm ERR! gyp ERR! find VS - not found
npm ERR! gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS You need to install the latest version of Visual Studio
npm ERR! gyp ERR! find VS including the "Desktop development with C++" workload.
npm ERR! gyp ERR! find VS For more information consult the documentation at:
npm ERR! gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Could not find any Visual Studio installation to use
npm ERR! gyp ERR! stack     at VisualStudioFinder.fail (C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:122:47)
npm ERR! gyp ERR! stack     at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:75:16
npm ERR! gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:363:14)   
npm ERR! gyp ERR! stack     at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:71:14
npm ERR! gyp ERR! stack     at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:384:16
npm ERR! gyp ERR! stack     at C:\project\UAVLogViewer\node_modules\node-gyp\lib\util.js:54:7
npm ERR! gyp ERR! stack     at C:\project\UAVLogViewer\node_modules\node-gyp\lib\util.js:33:16
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (node:child_process:427:5)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
npm ERR! gyp ERR! stack     at maybeClose (node:internal/child_process:1091:16)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:302:5)
npm ERR! gyp ERR! System Windows_NT 10.0.19045
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\project\\UAVLogViewer\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\project\UAVLogViewer\node_modules\deasync
npm ERR! gyp ERR! node -v v18.15.0
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok
npm ERR! Build failed

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Omar\AppData\Local\npm-cache\_logs\2023-09-10T16_01_03_837Z-debug-0.log

It gives npm errors please help me resolve it

Manouher, 2023-11-04:
> HEEEEEEEEEEEEEEEEEEEEELP

I installed it in a VM with Node.js v12, and it's working.

danikgrid, 2023-11-04:
> > HEEEEEEEEEEEEEEEEEEEEELP
> 
> I installed it in a VM with Node.js v12, and it's working.

can you tell us more about the installation process?

Manouher, 2023-11-05:
> > > HEEEEEEEEEEEEEEEEEEEEELP
> > 
> > 
> > I installed it in a VM with Node.js v12, and it's working.
> 
> can you tell us more about the installation process?

Can you share with me the error you have?

Manouher, 2023-11-05:
> npm ERR! code 1 npm ERR! path C:\project\UAVLogViewer\node_modules\deasync npm ERR! command failed npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c node ./build.js npm ERR! gyp info it worked [...]

npm install --global windows-build-tools

danikgrid, 2023-11-05:
> > npm ERR! code 1 npm ERR! path C:\project\UAVLogViewer\node_modules\deasync npm ERR! command failed npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c node ./build.js npm ERR! gyp info it worked if it ends with ok npm ERR! gyp info using node-gyp@8.4.1 npm ERR! gyp info using node@18.15.0 | win32 | x64 npm ERR! gyp info find Python using Python version 3.8.10 found at "C:\Users\Omar\AppData\Local\Programs\Python\Python38\python.exe" npm ERR! gyp ERR! find VS npm ERR! gyp ERR! find VS msvs_version not set from command line or npm config npm ERR! gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt npm ERR! gyp ERR! find VS checking VS2022 (17.6.33815.320) found at: npm ERR! gyp ERR! find VS "C:\Program Files\Microsoft Visual Studio\2022\Community" npm ERR! gyp ERR! find VS - found "Visual Studio C++ core features" npm ERR! gyp ERR! find VS - missing any VC++ toolset npm ERR! gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use npm ERR! gyp ERR! find VS looking for Visual Studio 2015 npm ERR! gyp ERR! find VS - not found npm ERR! gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8 npm ERR! gyp ERR! find VS npm ERR! gyp ERR! find VS ************************************************************** npm ERR! gyp ERR! find VS You need to install the latest version of Visual Studio npm ERR! gyp ERR! find VS including the "Desktop development with C++" workload. npm ERR! gyp ERR! find VS For more information consult the documentation at: npm ERR! gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows npm ERR! gyp ERR! find VS ************************************************************** npm ERR! gyp ERR! configure error npm ERR! gyp ERR! stack Error: Could not find any Visual Studio installation to use npm ERR! gyp ERR! stack at VisualStudioFinder.fail (C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:122:47) npm ERR! gyp ERR! stack at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:75:16 npm ERR! gyp ERR! stack at VisualStudioFinder.findVisualStudio2013 (C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:363:14) npm ERR! gyp ERR! stack at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:71:14 npm ERR! gyp ERR! stack at C:\project\UAVLogViewer\node_modules\node-gyp\lib\find-visualstudio.js:384:16 npm ERR! gyp ERR! stack at C:\project\UAVLogViewer\node_modules\node-gyp\lib\util.js:54:7 npm ERR! gyp ERR! stack at C:\project\UAVLogViewer\node_modules\node-gyp\lib\util.js:33:16 npm ERR! gyp ERR! stack at ChildProcess.exithandler (node:child_process:427:5) npm ERR! gyp ERR! stack at ChildProcess.emit (node:events:513:28) npm ERR! gyp ERR! stack at maybeClose (node:internal/child_process:1091:16) npm ERR! gyp ERR! stack at ChildProcess._handle.onexit (node:internal/child_process:302:5) npm ERR! gyp ERR! System Windows_NT 10.0.19045 npm ERR! gyp ERR! command "C:\Program Files\nodejs\node.exe" "C:\project\UAVLogViewer\node_modules\node-gyp\bin\node-gyp.js" "rebuild" npm ERR! gyp ERR! cwd C:\project\UAVLogViewer\node_modules\deasync npm ERR! gyp ERR! node -v v18.15.0 npm ERR! gyp ERR! node-gyp -v v8.4.1 npm ERR! gyp ERR! not ok npm ERR! Build failed
> > npm ERR! A complete log of this run can be found in: npm ERR! C:\Users\Omar\AppData\Local\npm-cache_logs\2023-09-10T16_01_03_837Z-debug-0.log
> > It gives npm errors please help me resolve it
> 
> try : `npm install --global windows-build-tools`

There's no change

danikgrid, 2023-11-05:
> > > > HEEEEEEEEEEEEEEEEEEEEELP
> > > 
> > > 
> > > I installed it in a VM with Node.js v12, and it's working.
> > 
> > 
> > can you tell us more about the installation process?
> 
> Can you share with me the error you have?

same

danikgrid, 2023-11-05:
> > > > HEEEEEEEEEEEEEEEEEEEEELP
> > > 
> > > 
> > > I installed it in a VM with Node.js v12, and it's working.
> > 
> > 
> > can you tell us more about the installation process?
> 
> Can you share with me the error you have?

https://gist.github.com/danikgrid/280eb8fcaef08eef05adc5a9e6

muirdok, 2024-11-20:
Same for Dockerfile:
`48.73 gyp ERR! configure error
48.73 gyp ERR! stack Error: Could not find any Python installation to use
48.73 gyp ERR! stack     at PythonFinder.fail (/usr/src/app/node_modules/node-gyp/lib/find-python.js:330:47)
48.73 gyp ERR! stack     at PythonFinder.runChecks (/usr/src/app/node_modules/node-gyp/lib/find-python.js:159:21)
48.73 gyp ERR! stack     at PythonFinder.<anonymous> (/usr/src/app/node_modules/node-gyp/lib/find-python.js:266:16)
48.73 gyp ERR! stack     at PythonFinder.execFileCallback (/usr/src/app/node_modules/node-gyp/lib/find-python.js:297:7)
48.73 gyp ERR! stack     at ChildProcess.exithandler (child_process.js:299:7)
48.73 gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
48.73 gyp ERR! stack     at maybeClose (internal/child_process.js:1022:16)
48.73 gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
48.73 gyp ERR! System Linux 5.15.0-119-generic`