ArduPilot/UAVLogViewer #404: 2.0 revamp list

Williangalvani, 2023-10-27:
- [ ] make main view dashboard-like. with multiple widgets (plot/3d/new stuff) that can be dragged around for repositioning
 - [ ] allow opening multiple logs
 - [ ] allow streaming data
 - [ ] allow multiple file formats
 - [ ] allow custom x/y plots

please comment any additional suggestions

Hwurzburg, 2023-10-27:
things I go back to MAVExplorer for:
-Magfit
-Paramchange
-overlay of messages on plots

-would be nice to export and import presets (primarily for backup
-export missions,fences,rally pts .....and perhaps show them on 3d view

Ultramarine1939-syujie, 2023-11-12:
I hope It can build on a new version of nodjs.I would get so many errors while I'm trying to install it on my laptop(nodejs 20-ubuntu20.04)

Ryanf55, 2024-03-24:
Tab complete or something similar for what's possible to plot.
[image]

IE: ATTITUDE -> `<tab tab>` results in a dropdown of the values that mavproxy console lists:
```
MANUAL> graph ATTITUDE.
ATTITUDE.pitch         ATTITUDE.pitchspeed    ATTITUDE.roll          ATTITUDE.rollspeed     ATTITUDE.time_boot_ms  ATTITUDE.yaw           ATTITUDE.yawspeed
```