ArduPilot/UAVLogViewer #407: Unable to view AOA.AOA

stw2nf, 2024-01-05:
[image]
The message is present and can be viewed through Mission Planner

VedantBhosle31, 2024-02-28:
Can you send that log file? 
So that we can recreate the error.

stw2nf, 2024-02-28:
Absolutely:
https://drive.google.com/file/d/1GtShN0oJSSshZmo3HvvmEhW9Hhi8usBr/view?usp=drive_link

It is there, I can pull from ardupilog and Mission Planner, for some reason not UAV Log Viewer though. I have tried Chrome and Edge as browsers.

axelisart, 2024-05-01:
Hi, just to say that this problem still occurs.

Lazersfly, 2024-06-24:
This is still a problem in both the production and beta log viewer websites. 

Happy to donate to the project if an option. This tool makes my post-flight analysis so much easier than the alternatives out there, would love to support the work to fix existing bugs and make the project better in general!

Williangalvani, 2024-07-06:
Hi, that should be fixed on master, can you give it a try?

Lazersfly, 2024-07-07:
AOA.AOA is working on plotbeta.ardupilot.org! Thank you!

closed 2024-07-07 by Williangalvani