ArduPilot/UAVLogViewer #414: docker build fails

clydemcqueen, 2024-04-17:
~~~
$ docker build --no-cache -t clydemcqueen/uavlogviewer .
[+] Building 41.1s (8/9)                                                                                                                                   docker:default
 => [internal] load build definition from Dockerfile                                                                                                                 0.0s
 => => transferring dockerfile: 343B                                                                                                                                 0.0s
 => [internal] load metadata for docker.io/library/node:12                                                                                                           0.3s
 => [internal] load .dockerignore                                                                                                                                    0.0s
 => => transferring context: 66B                                                                                                                                     0.0s
 => [1/5] FROM docker.io/library/node:12@sha256:01627afeb110b3054ba4a1405541ca095c8bfca1cb6f2be9479c767a2711879e                                                     0.0s
 => [internal] load build context                                                                                                                                    0.0s
 => => transferring context: 9.39kB                                                                                                                                  0.0s
 => CACHED [2/5] WORKDIR /usr/src/app                                                                                                                                0.0s
 => [3/5] COPY package*.json ./                                                                                                                                      0.1s
 => ERROR [4/5] RUN npm install                                                                                                                                     40.5s
------                                                                                                                                                                    
 > [4/5] RUN npm install:                                                                                                                                                 
0.659 npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but package-lock.json was generated for lockfileVersion@2. I'll try to do my best with it!                                                                                                                                                                 
36.06                                                                                                                                                                     
36.06 > deasync@0.1.27 install /usr/src/app/node_modules/deasync                                                                                                          
36.06 > node ./build.js
36.06 
36.11 `linux-x64-node-12` exists; testing
36.25 Binary is fine; exiting
36.73 
36.73 > node-sass@9.0.0 install /usr/src/app/node_modules/node-sass
36.73 > node scripts/install.js
36.73 
36.97 Downloading binary from https://github.com/sass/node-sass/releases/download/v9.0.0/linux-x64-72_binding.node
37.16 Download complete
37.16 Binary saved to /usr/src/app/node_modules/node-sass/vendor/linux-x64-72/binding.node
37.20 Caching binary to /root/.npm/node-sass/9.0.0/linux-x64-72_binding.node
37.32 
37.32 > @fortawesome/fontawesome-common-types@6.4.0 postinstall /usr/src/app/node_modules/@fortawesome/fontawesome-common-types
37.32 > node attribution.js
37.32 
37.36 Font Awesome Free 6.4.0 by @fontawesome - https://fontawesome.com
37.36 License - https://fontawesome.com/license/free (Icons: CC BY 4.0, Fonts: SIL OFL 1.1, Code: MIT License)
37.36 Copyright 2023 Fonticons, Inc.
37.36 
37.50 
37.50 > core-js@2.6.12 postinstall /usr/src/app/node_modules/core-js
37.50 > node -e "try{require('./postinstall')}catch(e){}"
37.50 
37.54 Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!
37.54 
37.54 The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
37.54 > https://opencollective.com/core-js 
37.54 > https://www.patreon.com/zloirock 
37.54 
37.54 Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)
37.54 
37.91 
37.91 > protobufjs@7.2.6 postinstall /usr/src/app/node_modules/protobufjs
37.91 > node scripts/postinstall
37.91 
38.13 
38.13 > es5-ext@0.10.62 postinstall /usr/src/app/node_modules/es5-ext
38.13 >  node -e "try{require('./_postinstall')}catch(e){}" || exit 0
38.13 
38.36 
38.36 > @fortawesome/free-solid-svg-icons@6.4.0 postinstall /usr/src/app/node_modules/@fortawesome/free-solid-svg-icons
38.36 > node attribution.js
38.36 
38.40 Font Awesome Free 6.4.0 by @fontawesome - https://fontawesome.com
38.40 License - https://fontawesome.com/license/free (Icons: CC BY 4.0, Fonts: SIL OFL 1.1, Code: MIT License)
38.40 Copyright 2023 Fonticons, Inc.
38.40 
38.41 
38.41 > bootstrap-vue@2.22.0 postinstall /usr/src/app/node_modules/bootstrap-vue
38.41 > opencollective || exit 0
[... 141 lines cut ...]
40.23 npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules/fsevents):
40.23 npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
40.23 
40.24 npm ERR! code ELIFECYCLE
40.24 npm ERR! errno 1
40.24 npm ERR! node-sass@9.0.0 postinstall: `node scripts/build.js`
40.24 npm ERR! Exit status 1
40.24 npm ERR! 
40.24 npm ERR! Failed at the node-sass@9.0.0 postinstall script.
40.24 npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
40.26 
40.26 npm ERR! A complete log of this run can be found in:
40.26 npm ERR!     /root/.npm/_logs/2024-04-17T19_44_28_576Z-debug.log
------
Dockerfile:11
--------------------
   9 |     COPY package*.json ./
  10 |     
  11 | >>> RUN npm install
  12 |     
  13 |     # Bundle app source
--------------------
ERROR: failed to solve: process "/bin/sh -c npm install" did not complete successfully: exit code: 1

~~~

yegawong, 2024-05-28:
try to 
FROM node:20

AspenEyers, 2024-06-03:
I had the same issue, modifying the docker file to use `FROM node:20` worked for me

bedaberner, 2025-06-04:
I fixed this in https://github.com/ArduPilot/UAVLogViewer/pull/431, this can be closed.