ArduPilot/UAVLogViewer #415: Downloaded data.csv does not seem to be correct

ntouev, 2024-04-28:
Downloading data to a csv file does not seem to be correct. 

- Plotting the downloaded data results to a graph slightly different than the one shown the UAV Log Viewer. See two graphs below.

**UAV Log Viewer plot**
[image]

**Using Excel to plot the downloaded data**
[image]

- Downloading the data with mavlogdump.py results to slightly different dataset.

**Discuss thread:** https://discuss.ardupilot.org/t/uav-log-viewer-downloaded-data-csv-problem/116881
**Tested Flight log:** https://drive.google.com/file/d/13SzHmCFXVaDOSkUoY7rSCYr1NIuQwrz_/view?usp=sharing

Williangalvani, 2024-07-06:
the issue here seems to be that we are just holding the last value instead of doing the linear interpolation that plotly does in chart. we are also re-sampling the logs at 100ms, as it is hard to decide what to do with the timestamp once you have multiple messages at different times being exported.
I'll take a look at it

Williangalvani, 2024-07-06:
Hi, could you try it again?
your log is no longer accessible, so I can't compare directly

ntouev, 2024-07-07:
Hi thanks for looking into that. Sure here is the log.