ArduPilot/UAVLogViewer #416: Cant display map data

kevinjones4642, 2024-05-06:
The icon in the upper right of the map screen, where I used to be able to select the map data source is missing. It's been a while since I used the viewer. Perhaps they just moved it, but I can't seem to find it.

[image]

Williangalvani, 2024-05-08:
Hi,
I can't replicate the issue. do you any errors in the browser console?
does it happen if you use the sample insted of loading your file?

kevinjones4642, 2024-05-09:
Thanks for the response. I get it with any log. The buttons on the upper
right side of the map window have changed, and I no longer see a button to
set the background map source. I've tried it on several PCs, and get the
same result. The 5 buttons allow me to: choose between 2D/3D, get
navigation instructions, center the vehicle, re-center the view, and close
the 3D view. It does read the log files, and I can make plots of parameters
in the file, I just can't see any terrain other than blue. I do not see any
error messages - just a slightly changed GUI.

--Kevin

kevinjones4642, 2024-05-09:
One last thing, I normally use Chrome, but for kicks I tried it in Edge,
and got the same thing.