ArduPilot/UAVLogViewer #420: doesn't open files

ValeriiaIZH, 2024-10-17:
Hello!
I want to open bin files after flight? but it doesn't open.
I used Opera, Edge and Google.
maybe it can be connected with discrepancy between autopilot and program versions ?
[image]
Could you help me?

podusenkovic, 2025-02-12:
same problem with infinite load screen
p.s: as it doesn't build with default Dockerfile from master - i switched it to node:20, may be it is the problem

Williangalvani, 2025-02-12:
hi, does it work on the online one at https://plotbeta.ardupilot.org?

Slimercorp, 2026-01-08:
It looks like internet issue. UAVLogViewer uses CESIUM, it can block some countries. Try VPN, it helped me