ArduPilot/UAVLogViewer #433: Cesium maps fail to load — Invalid Ion access token (401 Unauthorized)

olegpsu, 2025-05-16:
Hello ArduPilot team,

I'm encountering a critical issue when using https://plot.ardupilot.org/:
the Cesium 3D maps are no longer loading due to an invalid Cesium Ion access token.

🔍 Error details from browser console:
bash
Copy
Edit
Request URL: https://api.cesium.com/v1/assets/3954/endpoint?access_token=...
Status Code: 401 Unauthorized
Response Body:
{
  "code": "INVALID_TOKEN",
  "message": "Invalid access token"
}
This results in no terrain or map imagery being rendered, and causes related JavaScript features to break (e.g. 3D Viewer).

Williangalvani, 2025-05-17:
We had too many bing sessions and Cesium locked us out. I'm trying to sort this out.

jingpeng7527, 2025-05-22:
Hi, I’m encountering the same issue(401) when trying to run the project locally. Has this been resolved?

Williangalvani, 2025-05-22:
yes. update to master and run it with 
```VUE_APP_CESIUM_TOKEN=<Your cesium ion token> npm run dev```
or 
```docker run -e VUE_APP_CESIUM_TOKEN=<Your cesium ion token> -it -p 8080:8080 -v ${PWD}:/usr/src/app <your username>/uavlogviewer```

TillBerthold, 2025-05-24:
Thank you for fixing it!